suoto/hdl_checker #35: Helper tools to autogenerate PRJ config from other file formats

abyszuk, 2017-02-08:
Hello,

First of all, I'd like to thank you for vim-hdl and hdlcc. They are great tools and have a lot potential.
I think a useful addition would be to add simple tools which would automatically generate PRJ file from other project files e.g. Vivado *.XPR projects or *.XISE projects.
My rationale is that for large projects typing the PRJ configuration by hand can be a nightmare.
Even worse, in my job I have to frequently switch between many projects, which effectively prevents me from using hdlcc on a daily basis.

suoto, 2017-02-10:
Hi abyszuk, thanks for the feedback. Great to hear you find these tools useful :)

And yes, unfortunately setting up the project is not trivial. 

Supporting Vivado, ISE, etc configuration files natively or at least providing some helper tools is a good idea. In the meantime, you might want to use a script to search for RTL (VHDL, Verilog and SystemVerilog) files and generate the PRJ file. You can get it here: [create_hdlcc_project.py](https://gist.github.com/suoto/f687839cfabf02e91c5740d530c1fc8c)

abyszuk, 2017-02-13:
Great, thanks!

This will definitely come in handy.

suoto, 2017-03-10:
Provided there are no further requests I'm closing this. Please let me know if you need any further help!

closed 2017-03-10 by suoto