suoto/hdl_checker #41: Maintaining project file

m-kru, 2018-12-10:
@suoto I find your project really interesting, you had great idea!

There is only one thing that bothers me. Do you use hdlcc on a daily basis?
It looks like project files for hdlcc describe FPGA project file structure. Usually RTL file structure of FPGA project is also described in different files (for example TCL scripts or project files generated by vendors IDEs). Maintaining both is inconvenient and it is easy to forget to update hdlcc project file. Would you be so nice to explain how you handle this issue?

suoto, 2018-12-10:
Hey, thanks for the words :)

So, fully agree that maintaining a separate file is a hassle... Lately I was using [this Python script](https://gist.github.com/suoto/f687839cfabf02e91c5740d530c1fc8c) to regenerate the project file but that's a workaround really.

Maybe supporting vendor's formats (and assuming sensible defaults) is the way to go. The list of files is probably easy to extract from Vivado's/ISE/Quartus project files, just need a way to add specific stuff as well (for example, I set VHDL 93 for things that will be synthesized and VHDL 2008 for testbenches).

m-kru, 2018-12-11:
Extracting the list of files from Vivado's/ISE/Quartus project files is one potential solution.
For me, the drawback of this idea is that I often work in non-project mode and I use [vextproj](https://github.com/wzab/vextproj) to easily recreate the project. 

I wonder if it wouldn't be better to create some common format that would describe FPGA project file structure/hierarchy. Then we would need scripts for translating this to different tools, but there would be only one structure defining the project. What do you think?

suoto, 2019-01-04:
I think there might be some projects out there that try to be this entry point for multiple EDA vendor tools, have a look at https://github.com/olofk/fusesoc (I just read the README, haven't tried just yet...).

Currently, updating the files/flags database is triggered by changing the project file, but maybe hdlcc could have a way to make it easier to maintain. I can think of 2 ways for this to work:

I can think of 2 ways for this to work:

1. Provide an API in Vim or Python for the user to define the project. It would be basically an add file/flag method but it could also have a callback for hdlcc to check if a refresh is needed. Downside is the user would need to actually code this
1. Add support for wildcards in the project file, so that the user only needs to specify some paths and hdlcc would then run that regularly and trigger recompiling stuff if things have changed

Does this sounds reasonable in you workflow @m-kru?

m-kru, 2019-01-04:
@suoto I have also discovered FuseSoC recently but I haven't tried it yet.
I was thinking about API in Python. There are already a lot of tools for HDL development in Python and it is just easier to work when you use the same language for different tools.

suoto, 2019-02-17:
I'm working on a command to vim-hdl to create/update the project file. The idea is the user runs `VimhdlCreateProjectFile` to create the project file and review the file afterwards.

I'll separate the core stuff in a base class, so implementations only need to add sources and all the base class then takes care of translating that into a proper project file.

The only implementation I'm working for now just finds sources and include paths on the directories passed as argument (i.e., `:VimhdlCreateProjectFile ./project/src  ./project/test`). Depending on how this works, it shouldn't be too hard to add implementations that parse Vivado, Quartus, etc.

suoto, 2019-07-31:
Going to try to implement an auto discovery method that parses sources and tries to work out the project's layout.

The config file will hint to sorting out ambiguity and specifying build flag, but ideally for small and maybe mid size projects a config file won't be needed at all.

rdburns, 2019-10-24:
I am also interested in a common definition of source files, in my case originally to unify the sources between Questa sims and non-project Vivado. But I am also interested in my editor (at the moment spacemacs, but with LSP, hopefully anything) understanding the same files.

I looked into cmake and meson a bit for this purpose, since apparently cmake already has a facility to write visual studio solution files, or run as a server. I don't fully understand all that. Using either of those would require adding custom builders. This project basically does this for cmake but I have not tried it: https://github.com/tymonx/logic

suoto, 2019-10-24:
Yeah, that's a question with no good answer imo. Each tool (including this one) uses a different config method. I think HDL Checker is pretty usable as is, one can use it without a config file if t  he project allows or use an incomplete one, listing only files, or listing files and defining libraries for some or using one with all the info.

suoto, 2019-10-25:
Would be possible to support project files from Vivado, Quartus, etc, although integrating that on the testing pipeline needs a bit a thought

suoto, 2020-01-14:
Going to close this for now as the auto discovery of files and libraries seems to be working well enough. If there's any specific questions/issues feel free to open a new issue :+1:

closed 2020-01-14 by suoto