suoto/hdl_checker #42: Language server protocol support

abyszuk, 2019-01-24:
Language server protocol is all the rage these days.
I find hdlcc to be very useful VHDL code checker, but limited only to vim. LSP would allow it to be used also in VS Code, Atom and other editors.
Would such a task be feasible? Do you think it's worth the effort?

kazooiebombchu, 2019-01-28:
I would also like to see this implemented. I love to use this with vim so far and haven't seen similar tool that compiles the sources on save. Or have I missed something?

suoto, 2019-02-01:
Had never heard about LSP before to be honest :)

The LSP defines quite a lot of interesting things, hdlcc could possibly implement the Diagnostics interface, but I need to check if/how LSP supports configuration (like the path to the hdlcc config file) and if the protocol requires servers to send results as you type (i.e., without having to save the file; should be doable, just a bit more work really). 

@kazooiebombchu if hdlcc can locate a compiler to use, it should compile the sources you save and its dependencies using a hidden folder as workspace. You can always use those results if you need, although if you're actually running simulations, I'd recommend [VUnit](https://github.com/VUnit/vunit) :)

abyszuk, 2019-02-05:
AFAIK LSP supports sending both file diff and whole file. Simpler servers implement only whole-file mode on file save, anything more is optional.
LSP supports passing path to configuration file or top directory. This is called rootUri.
This is how such configuration looks like for another plugin working with vim-lsp:
https://github.com/prabirshrestha/vim-lsp/wiki/Servers-VHDL

suoto, 2019-07-17:
LSP support should be working on the [lsp_support](https://github.com/suoto/hdlcc/tree/lsp_support) branch - at least seems to work fine with [w0rp/ale](https://github.com/w0rp/ale), but still need to update the readme with changes. The Vim side of things is on a similar state, except it still need proper testing.

(took a while to get this properly working, had other stuff going on)

suoto, 2019-07-31:
Forgot to close this. LSP has been integrated into master. Will update the readme with info on how to use it soon.

closed 2019-07-31 by suoto