suoto/hdl_checker #51: Project file detection

skaupper, 2019-10-02:
First of all, nice work you did here.

Could you elaborate how hdlcc is detecting its project files? I was testing your vim-hdl example but couldn't figure it out completely

suoto, 2019-10-03:
Hey, thanks! I haven't updated https://github.com/suoto/vim-hdl-examples just yet, partly because that one uses Syntastic.

When using LSP mode, HDL Checker will look for a file named `.hdl_checker.config` (or `_hdl_checker.config` if on Windows) in the root path of the workspace. If the setup you're using is the same as the log you posted in #52, that would be `/home/skaupper/repos/vhdl-lsp-client/vhdl_test`, so the config file should be `/home/skaupper/repos/vhdl-lsp-client/vhdl_test/.hdl_checker.config`.

HDL Checker will try interpret as JSON file first and, if that fails, will try to parse using the legacy prj format (they're described in the readme).

Let me know if that helps!

skaupper, 2019-10-03:
Originally, I was testing the `vim-hdl-example` without a valid config and was a little confused why it was working. But I figured to the zero conf usage described in the [wiki](https://github.com/suoto/hdl_checker/wiki/Zero-Conf-implementation-details).  

While using LSP mode on its own (without `vim-hdl` and in my own project) I encountered the same fallback and your explanation is indeed helpful there.  
Maybe it is worth clearifying the config behaviour (default names, `HDL_CHECKER_DEFAULT_PROJECT_FILE`, a link to the zero conf page, ...) in the readme?

suoto, 2019-10-03:
Absolutely agree, documentation is lagging really!

suoto, 2019-10-10:
Added relevant info to the wiki: https://github.com/suoto/hdl_checker/wiki/Setting-up-a-project

The zero conf page was more of a scratch pad for myself, will get to it in due time :)

closed 2019-10-10 by suoto