suoto/hdl_checker #52: Wrong dependency error

skaupper, 2019-10-03:
While fiddling around with the LSP version I got some multiple definition warnings by the HDL Checker.

The code I used:
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package very_common_pkg is
end package;

package body very_common_pkg is
end package body;
```

When I removed (and later added) the last semicolon I got the warning
`Dependency 'very_common_pkg' (library=None) has 2 definitions (files are "/home/skaupper/repos/vhdl-lsp-client/vhdl_test/lib1/foo.vhd", "/home/skaupper/repos/vhdl-lsp-client/vhdl_test/lib1/foo.vhd"). The selected option may not be the correct one`
and at the same moment
`Dependency 'very_common_pkg' (library=None) has 2 definitions (files are "/home/skaupper/repos/vhdl-lsp-client/vhdl_test/lib1/foo.vhd", "/tmp/tmp072shbl8.vhd"). The selected option may not be the correct one`

Let me know if you need further information.

[hdl_checker_log_pid44064.log](https://github.com/suoto/hdl_checker/files/3685176/hdl_checker_log_pid44064.log)

suoto, 2019-10-03:
Yes, thanks for reporting that. Saw this yesterday only, will get a fix soon

skaupper, 2019-10-03:
Nice to hear that, looking forward to it.

closed 2019-10-08 by suoto