suoto/hdl_checker #54: Support for regex in sources

chrbirks, 2019-10-16:
Hi
Is there, or do you have plans for, support for specifying multiple source files using regex or something similar? For example to include all .vhd and .vhdl files in all subdirectories:
```
{
   "sources": [
      "src/**/*.vhd*"
   ]
}
```

chrbirks, 2019-10-17:
Okay I wasn't aware that adding sources to the config file while the server is running was possible. So yes, indexing new files should be handled in a nice way somehow. The LSP server I currently use for VHDL cannot handling adding/removing source files while running so that hasn't been a problem.

closed 2019-10-22 by suoto

suoto, 2019-10-22:
Support for specifying using Python glob, which allows patterns like `/project/foo/*.vhd` and, in case of Python 3, allows recursive inclusion with `/project/**/*.vhd` (will include all `.vhd` files inside `/project/`). More info on how glob behaves, have a look at https://docs.python.org/3/library/glob.html.

The server won't automatically search for files, to trigger that you can (1) remove HDL Checker work folder (`<root_path>/.hdl_checker/` on Unix, `<root_path>\\_hdl_checker\\` on Win) or (2) change the config file timestamp.

Thanks for reporting, let me know if there's any other issue! :+1:

chrbirks, 2019-10-22:
Thanks I'll give et a try in the next release.

commit aade2e32 (2019-10-22): Adds support for glob on config files (closes #54)