suoto/hdl_checker #56: How to add unisim library?

fransschreuder, 2019-10-21:
Hello, thanks for your great work, it all looks promising, especially with the vs code plugin. The only thing I am not getting is how to add specific arguments to pass to ghdl (like library search paths and ieee=synopsys

suoto, 2019-10-21:
Hi @fransschreuder, thanks for the feedback!

Settings on the VS Code client are only for the client, not the server.

Setting specific compile flags is done via config file (file named `.hdl_checker.config` if on Linux or `_hdl_checker.config` if on Windows in the root of your project). To pass the `--ieee=synopsys` argument for every GHDL call, add the following to that file

```json
{
    "vhdl": {
        "flags": {
            "global": ["--ieee=synopsys"]
        }
    }
}
```

I'm not a GHDL expert, but it seems you can use [`--PREFIX=<path>`](https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#cmdoption-ghdl-prefix) to configure where ieee and std libraries are.

More detail on the config file fields can be found [on the Wiki](https://github.com/suoto/hdl_checker/wiki/Setting-up-a-project#using-json-file)

As a side note: if you were relying on HDL Checker searching the files, issue #53 will apply (HDL Checker will *not* search for files if it can find a config file, even if it has no sources). To work around that, you'd also need to to add the paths manually (again, the wiki has info on that).

fransschreuder, 2019-10-21:
Hi Suoto,
Thanks for the great answer. I missed the wiki, only checked README.md. A link to the wiki in the readme would be handy, unless I missed that as well.
Thanks and regards.
Frans

Ps. Can I donate anywhere?

closed 2019-10-21 by fransschreuder

suoto, 2019-10-22:
It's alright, docs are still WIP, there's (now) a link to the wiki on the readme, but the wiki itself is not yet properly organized...

I don't have donation setup, appreciate very much the intention though :)