suoto/hdl_checker #60: Paths were not built

chrbirks, 2019-11-04:
I have a config that includes a single file as source but the hdl_checker log file says the file path is not built and missing simicolons etc. are not reported. Is is possible to find out what the problem is?

I'm starting hdl_checker in lsp mode through lsp-mode for Emacs and the full log is below

```
INFO    | 20:14:29 | hdl_checker.server @ startServer():219 MainThread |	Starting server. Our PID is 18461, no parent PID to attach to. Version string for hdl_checker is '0.6.9'
INFO    | 20:14:29 | pyls.python_ls @ start_io_lang_server():85 MainThread |	Starting HdlCheckerLanguageServer IO language server
WARNING | 20:14:29 | pyls.config.config @ __init__():55 MainThread |	Failed to load pyls entry point 'autopep8': No module named 'autopep8'
WARNING | 20:14:29 | pyls.config.config @ __init__():55 MainThread |	Failed to load pyls entry point 'pydocstyle': No module named 'pydocstyle'
WARNING | 20:14:29 | pyls.config.config @ __init__():55 MainThread |	Failed to load pyls entry point 'pylint': No module named 'pylint'
WARNING | 20:14:29 | pyls.config.config @ __init__():55 MainThread |	Failed to load pyls entry point 'rope_completion': No module named 'rope'
WARNING | 20:14:29 | pyls.config.config @ __init__():55 MainThread |	Failed to load pyls entry point 'rope_rename': No module named 'rope'
WARNING | 20:14:29 | pyls.config.config @ __init__():55 MainThread |	Failed to load pyls entry point 'yapf': No module named 'yapf'
INFO    | 20:14:29 | hdl_checker.utils @ wrapper():362 MainThread |	m_initialize((), {'capabilities': {'textDocument': {'codeAction': {'codeActionLiteralSupport': {'codeActionKind': {'valueSet': ['',
                                                                                                               'quickfix',
                                                                                                               'refactor',
                                                                                                               'refactor.extract',
                                                                                                               'refactor.inline',
                                                                                                               'refactor.rewrite',
                                                                                                               'source',
                                                                                                               'source.organizeImports']}},
                                                  'dynamicRegistration': True},
                                   'completion': {'completionItem': {'documentationFormat': ['markdown'],
                                                                     'snippetSupport': True},
                                                  'contextSupport': True},
                                   'declaration': {'linkSupport': True},
                                   'definition': {'linkSupport': True},
                                   'documentLink': {'dynamicRegistration': True},
                                   'documentSymbol': {'hierarchicalDocumentSymbolSupport': True,
                                                      'symbolKind': {'valueSet': [1,
                                                                                  2,
                                                                                  3,
                                                                                  4,
                                                                                  5,
                                                                                  6,
                                                                                  7,
                                                                                  8,
                                                                                  9,
                                                                                  10,
                                                                                  11,
                                                                                  12,
                                                                                  13,
                                                                                  14,
                                                                                  15,
                                                                                  16,
                                                                                  17,
                                                                                  18,
                                                                                  19,
                                                                                  20,
                                                                                  21,
                                                                                  22,
                                                                                  23,
                                                                                  24,
                                                                                  25,
                                                                                  26]}},
                                   'foldingRange': {'dynamicRegistration': True,
                                                    'lineFoldingOnly': False,
                                                    'rangeLimit': None},
                                   'formatting': {'dynamicRegistration': True},
                                   'hover': {'contentFormat': ['markdown',
                                                               'plaintext']},
                                   'implementation': {'linkSupport': True},
                                   'rangeFormatting': {'dynamicRegistration': True},
                                   'rename': {'dynamicRegistration': True},
                                   'semanticHighlightingCapabilities': {'semanticHighlighting': None},
                                   'signatureHelp': {'signatureInformation': {'parameterInformation': {'labelOffsetSupport': True}}},
                                   'synchronization': {'didSave': True,
                                                       'willSave': True,
                                                       'willSaveWaitUntil': True},
                                   'typeDefinition': {'linkSupport': True}},
[... 43 lines cut ...]
INFO    | 20:14:36 | hdl_checker.lsp @ _getDiags():331 Thread-4 |	Linting Path('/home/cbs/projects/train/train/src/hdl/common.vhd') (saved=True)
INFO    | 20:14:36 | hdl_checker.lsp @ lint():295 Thread-5 |	linting: file:///home/cbs/projects/train/train/src/hdl/common_top.vhd
INFO    | 20:14:36 | hdl_checker.lsp @ _getDiags():331 Thread-5 |	Linting Path('/home/cbs/projects/train/train/src/hdl/common_top.vhd') (saved=True)
INFO    | 20:14:36 | hdl_checker.lsp @ lint():295 Thread-7 |	linting: file:///home/cbs/projects/train/train/src/hdl/registers_pkg.vhd
INFO    | 20:14:36 | hdl_checker.lsp @ _getDiags():331 Thread-7 |	Linting Path('/home/cbs/projects/train/train/src/hdl/registers_pkg.vhd') (saved=True)
INFO    | 20:14:36 | hdl_checker.lsp @ lint():295 Thread-8 |	linting: file:///home/cbs/projects/train/train/src/hdl/revision.vhd
INFO    | 20:14:36 | hdl_checker.lsp @ _getDiags():331 Thread-8 |	Linting Path('/home/cbs/projects/train/train/src/hdl/revision.vhd') (saved=True)
INFO    | 20:14:36 | hdl_checker.lsp @ lint():295 Thread-6 |	linting: file:///home/cbs/projects/train/train/src/hdl/fb_capabilities.vhd
INFO    | 20:14:36 | hdl_checker.lsp @ _getDiags():331 Thread-6 |	Linting Path('/home/cbs/projects/train/train/src/hdl/fb_capabilities.vhd') (saved=True)
INFO    | 20:14:36 | hdl_checker.database @ addSource():151 Thread-4 |	Adding /home/cbs/projects/train/train/src/hdl/revision.vhd, library=work, flags=(single=('--std=08',), dependencies=('--std=08',))
INFO    | 20:14:36 | hdl_checker.database @ _updatePathLibrary():324 Thread-4 |	Setting library for '/home/cbs/projects/train/train/src/hdl/common.vhd' to 'not_in_project'
[... cut ...]
INFO    | 20:14:37 | hdl_checker.database @ getBuildSequence():766 Thread-5 |	Nothing more to do after 0 steps
INFO    | 20:14:37 | hdl_checker.builders.fallback @ build():420 Thread-5 |	Forcing build of /home/cbs/projects/train/train/src/hdl/common_top.vhd
INFO    | 20:14:37 | hdl_checker.database @ getBuildSequence():766 Thread-8 |	Nothing more to do after 0 steps
INFO    | 20:14:37 | hdl_checker.builders.fallback @ build():420 Thread-8 |	Forcing build of /home/cbs/projects/train/train/src/hdl/revision.vhd
```

suoto, 2019-11-04:
Can you run `ghdl --version` on the command line?

From the logs you're trying to use GHDL but it seems that GHDL executable is not in your PATH, so HDL Checker will use the fallback compiler (which does nothing). The paths not built is likely caused by this as well; HDL Checker will consider a dependency satisfied when it can find a file defining it or when the compiler has it built-in (or pre compiled). Because GHDL couldn't be found, the most basic libraries (std, ieee, etc) will also be missing

chrbirks, 2019-11-04:
Ah the problem was that I was running it on my work computer which didn't have ghdl installed. But now it reports another file not being built even though it's not part of the config file: 
`WARNING | 21:13:39 | hdl_checker.database @ getBuildSequence():763 Thread-4 |   1 paths were not built: ['/home/cbs/projects/train/train/src/hdl/fb_capabilities.vhd']`
Can it be a cashing issue from a previous version of the config file that included all files in the directory?

```
{
  "sources": [
    ["/home/cbs/projects/train/train/src/hdl/revision.vhd", {"library": "work"}]
  ],

  "vhdl": {
      "flags": {
        "single": [],
        "dependencies": [],
        "global": ["--std=08"]
      }
    },
    "verilog": {
      "flags": {
        "single": [],
        "dependencies": [],
        "global": []
      }
    },
    "systemverilog": {
      "flags": {
        "single": [],
        "dependencies": [],
        "global": []
      }
    }
}
```

suoto, 2019-11-04:
Files not listed in the config file will be ignored until they're opened, after that they'll be checked as usual, though they should get an extra warning reporting this.

As for the cache, it should be invalidated when the config file changes. If you're unsure, you can delete the `.hdl_checker/` folder; it will be recreated when needed :)

chrbirks, 2019-11-05:
I think the warning about `fb_capabilities.vhd` was because it was opened automatically in the background by Emacs but not included as a source in the config file. I don't see any problems now except for the warnings about missing Python modules.

So the main problem was that ghdl was not installed :-)

suoto, 2019-11-05:
Great, glad that is solved :) 

Unresolved dependencies preventing a file being compiled sometimes confuse me as well, will have a look at reporting that so we don't need to crawl logs

Going to close this, feel free to reopen of there's any other question!

closed 2019-11-05 by suoto