suoto/hdl_checker #62: Added to Emacs lsp-mode

chrbirks, 2019-11-10:
Just a FYI. I added support for HDL Checker to the Emacs package `lsp-mode`: https://github.com/emacs-lsp/lsp-mode/pull/1167
So it is now easy to just add the following to your Emacs config. Setting the server path is only required if the server binary is not in your PATH
```
(require 'use-package)
(setq lsp-vhdl-server-path "~/.local/bin/hdl_checker")
(custom-set-variables
 '(lsp-vhdl-server 'hdl-checker))
(use-package lsp-mode
	     :config (add-hook 'vhdl-mode-hook 'lsp))
```

suoto, 2019-11-10:
That's great, thanks! :)

Would be nice to add this to the [editors support section on the readme](https://github.com/suoto/hdl_checker#editor-support), but I'm not familiar with emacs at all. Maybe adding this bit and referencing emacs-lsp/lsp-mode enough?

> ```
> (require 'use-package)
> (setq lsp-vhdl-server-path "~/.local/bin/hdl_checker")
> (custom-set-variables
>  '(lsp-vhdl-server 'hdl-checker))
> (use-package lsp-mode
> 	     :config (add-hook 'vhdl-mode-hook 'lsp))
> ```

chrbirks, 2019-11-10:
Yes that should be enough. It requires that `use-package` is already installed but it's already pretty common.

chrbirks, 2019-11-10:
Well the code was just reverted but it will be back again once I have fixed a problem :)

suoto, 2019-11-10:
Ok, no worries. The readme has the section marked as "soon", will remove once the emacs-lsp is OK!

closed 2019-11-10 by suoto

chrbirks, 2019-11-14:
The problem has been fixed and merged into the lsp-mode package again.

commit 93ff44d6 (2019-11-10): Added setup info for emacs-lsp to the readme (closes #62)
commit b4b8db82 (2019-11-10): Added setup info for emacs-lsp to the readme (closes #62)