suoto/hdl_checker #64: Project files

rafaelnp, 2019-11-19:
I saw in the hdl_checker wiki the page to create a project, but I could not file where (directory) the project file should be saved ($HOME, root directory of the project, etc..) and how the project file should be named. Could please clarify these two questions?  Thanks. :)

rafaelnp, 2019-11-19:
I found the filename at the bottom of the wiki page, my bad. But regarding the directory/path, it is still not clear to me.

suoto, 2019-11-20:
The `HDL_CHECKER_DEFAULT_PROJECT_FILE` environment variable refers only to the name of the config file, HDL Checker will look for it in the root of the workspace, which is set by the client (your editor).

Usually that would be the project's root directory (e.g, on a git repo, that'd be the result of `git rev-parse --show-toplevel`, that's where I usually keep it).

rafaelnp, 2019-11-20:
OK, thanks for the clarification.

closed 2019-11-20 by rafaelnp