suoto/hdl_checker #73: hdl checker on termux

StephanBeer, 2019-11-29:
I am trying to get hdl checker working on/with termux with neovim and coc, but when installing with "pip install hdl-checker --upgrade " i get the following error and lsp isnt starting

~~~
 ⌘ ✘  hdl_checker --version                        ~
Traceback (most recent call last):
  File "/data/data/com.termux/files/usr/bin/hdl_checker", line 11, in <module>
    load_entry_point('hdl-checker==0.6.11', 'console_scripts', 'hdl_checker')()
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 2852, in load_entry_point
    return ep.load()
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 2443, in load
    return self.resolve()
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 2449, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/hdl_checker/server.py", line 29, in <module>
    from pyls.python_ls import start_io_lang_server  # type: ignore
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/pyls/python_ls.py", line 10, in <module>
    from pyls_jsonrpc.streams import JsonRpcStreamReader, JsonRpcStreamWriter
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/pyls_jsonrpc/streams.py", line 5, in <module>
    import ujson as json
ImportError: dlopen failed: cannot locate symbol "Buffer_AppendShortHexUnchecked" referenced by "/data/data/com.termux/files/usr/lib/python3.8/site-packages/ujson.cpython-38.so"...
 9  ⌘ ✘ 
~~~

suoto, 2019-11-29:
Can you try installing Python 3.8 dev package? (that'd be the `python3.8-dev` package on Debian)

Or could you try installing ujson manually to make sure it succeeds?

That's because one of HDL Checker dependencies has moved to use ujson instead of the regular json module, which is compiled when installing. Hopefully #57 is more stable in this regard.

StephanBeer, 2019-12-23:
Thank you for your reply. There is no python-dev version in the Termux package repository so i installed ujson manually which succeeded. Still i get the error when i start the hdl_checker.

StephanBeer, 2019-12-23:
~~~
pip install ujson
Requirement already satisfied: ujson in /data/data/com.termux/files/usr/lib/python3.8/site-packages (1.35)
~~~

StephanBeer, 2019-12-23:
~~~
Traceback (most recent call last):
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 583, in _build_master
    ws.require(__requires__)
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 900, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 791, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (requests 2.22.0 (/data/data/com.termux/files/usr/lib/python3.8/site-packages), Requirement.parse('requests==2.20.0'), {'hdl-checker'})

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/data/data/com.termux/files/usr/bin/hdl_checker", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3252, in <module>
    def _initialize_master_working_set():
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3235, in _call_aside
    f(*args, **kwargs)
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3264, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3235, in _call_aside
    f(*args, **kwargs)
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3264, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 585, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 598, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 786, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'requests==2.20.0' distribution was not found and is required by hdl-checker
~~~

StephanBeer, 2019-12-23:
Now i get at least the version number
~~~   master  ⌘ ✘  hdl_checker --version
0.6.12
~~~

StephanBeer, 2019-12-23:
Starting hdl_checker with 'hdl_checker --lsp' still throws an error 
~~~
Traceback (most recent call last):
  File "/data/data/com.termux/files/usr/bin/hdl_checker", line 11, in <module>
    load_entry_point('hdl-checker==0.6.12', 'console_scripts', 'hdl_checker')()
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/hdl_checker/server.py", line 225, in main
    return run(parseArguments())
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/hdl_checker/server.py", line 180, in run
    _setupPipeRedirection(None if args.lsp else args.stdout, args.stderr)
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/hdl_checker/server.py", line 146, in _setupPipeRedirection
    sys.stderr = openForStdHandle(stderr)
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/hdl_checker/server.py", line 138, in openForStdHandle
    return open(filepath, mode="w", buffering=1)
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/hdl_checker_stderr_pid21690.log'
~~~

StephanBeer, 2019-12-23:
hdl_checker --help gives...
~~~
31  ✘  hdl_checker --help             1s  ~/c/verilog
usage: hdl_checker [-h] [--host HOST] [--port PORT]
                   [--lsp]
                   [--attach-to-pid ATTACH_TO_PID]
                   [--log-level LOG_LEVEL]
                   [--log-stream LOG_STREAM]
                   [--stdout STDOUT] [--stderr STDERR]
                   [--version]

optional arguments:
  -h, --help            show this help message and exit
  --host HOST           [HTTP] Host to serve
  --port PORT           [HTTP] Port to serve
  --lsp                 Starts the server in LSP mode.
                        Defaults to false
  --attach-to-pid ATTACH_TO_PID
                        [HTTP, LSP] Stops the server if
                        given PID is not active
  --log-level LOG_LEVEL
                        [HTTP, LSP] Logging level
  --log-stream LOG_STREAM
                        [HTTP, LSP] Log file, defaults to
                        stdout when in HTTP or a
                        temporary file named
                        hdl_checker_log_pid<PID>.log when
                        in LSP mode
  --stdout STDOUT       [HTTP] File to redirect stdout
                        to. Defaults to a temporary file
                        named
                        hdl_checker_stdout_pid<PID>.log
  --stderr STDERR       [HTTP] File to redirect stdout
                        to. Defaults to a temporary file
                        named
                        hdl_checker_stderr_pid<PID>.log
  --version, -V         Prints hdl_checker version and
                        exit
~~~

StephanBeer, 2019-12-23:
On Termux there is no /tmp directory, only /usr/tmp ...so logging file cant be written.
In NeoVim with CoC and Ale, Hdl Checker seems to be working.

suoto, 2020-01-22:
When starting the server, use `--log-stream NONE` (in uppercase) and/or `--stderr NONE` to disable logging and/or writing to stderr respectively.

Thanks for the patience!

commit 415335ab (2020-01-22): Use --log-stream NONE and/or --sterr NONE to disable logging, stderr

Closes #73
commit 3b7b973f (2020-01-22): Use --log-stream NONE and/or --sterr NONE to disable logging, stderr

Closes #73
closed 2020-01-22 by suoto