suoto/hdl_checker #75: Type DefaultDict cannot be instantiated;

kdurant, 2019-12-13:
```
[object Object]
[Error - 5:03:20] Request textDocument/hover failed.
  Message: TypeError: Type DefaultDict cannot be instantiated; use collections.defaultdict() instead
  Code: -32602 
```

suoto, 2019-12-13:
Which Python version and OS are you getting this?

kdurant, 2019-12-15:
python3.6
win10

kdurant, 2019-12-15:
python3.6
win10

suoto, 2019-12-15:
Can't tell much without logs. Might be something to do with the cache state; have you tried cleaning up the the `_hdl_checker` folder in the project's root?

kdurant, 2019-12-16:
Hdl_checker without settings, only install `HDL Checker LSP Client` in vscode
Project without `_hdl_checker` folder 
Only one file in project

suoto, 2020-01-14:
Which HDL Checker version are you using? You can check it via `hdl_checker -V`

suoto, 2020-01-29:
Closing due to inactivity, please reopen if/when the requested info is provided.

suoto, 2020-02-05:
This issue seems to happen on Python 3.5, not Python 3.6, will investigate further, but if adding Python 3.5 to the CI tests is too much of a hassle it won't be worth it since its end of life is only 7 months away.

Tcenova, 2020-02-06:
Can confirm this occurs on Python 3.5 on Ubuntu 16.04. It seems that the typing library was added to Python 3.5 and the pip package has no impact according to: https://pypi.org/project/typing/. It must be that the older version of the typing library did not support the instantiation of a DefaultDict.

suoto, 2020-02-07:
Might be, I've tested with Python 3.5 locally and didn't get this so it's not really Python version specific indeed.

In any case, I'll  remove uses of DefaultDict to fix this issue, feel free to reopen this or open another issue if there's other problems :+1:

commit 3462f298 (2020-02-07): Removing uses of typing.DefaultDict (fixes #75)
closed 2020-02-07 by suoto