suoto/hdl_checker #76: What should be the name of the .json file?

antoinemadec, 2019-12-16:
First, thanks for this plugin!

The default recursive method says my project is too big and that I should pass a .json file to point to the right files.

I followed this section of the Wiki: https://github.com/suoto/hdl_checker/wiki/HOWTO:-Setting-up-a-project#using-json-file

However, I don't know how to call the **json** file. I tried **project.json** but it didn't work.
**Could you please tell me how to name it?**

For info I am using **nvim + coc.nvim**.
Thanks in advance!

suoto, 2019-12-17:
Thanks!

The default project file is `.hdl_checker.config` if on Linux/Mac or `_hdl_checker.config` if on Windows (you can change that by setting the `HDL_CHECKER_DEFAULT_PROJECT_FILE` environment variable).

By the way, a project file helps avoiding parsing irrelevant files but if parsing can still take a bit, which will trigger the same warning, but opening the project the 2nd time should be faster.

Let me know if that works!

antoinemadec, 2019-12-17:
@suoto , **hdl_checker** is now picking my .config file, but it does not found any definition.
I will debug it further later if I have time.
Thanks again!

closed 2019-12-17 by antoinemadec

suoto, 2019-12-17:
Nice!

By the way, not all definitions will be found, only those relevant to defining the compilation order such as packages (`use work.some_pkg.all`) and named design units (`foo : entity work.some_entity`). This is not stated anywhere I think but probably should...!