suoto/hdl_checker #83: No module named 'six' on attempted launch

DeeeeLAN, 2020-08-06:
I created a python 3.6.4 venv and installed hdl_checker. I then attempted to check the version and was hit with the following error:

```
$ hdl_checker -V
Traceback (most recent call last):
  File
"/home/data/verification/users/millsd/workspace/uvm_mk_tb_pol/test_venv/bin/hdl_chec\
ker", line 11, in <module>
    load_entry_point('hdl-checker==0.6.13', 'console_scripts', 'hdl_checker')()
  File
"/home/data/verification/users/millsd/workspace/uvm_mk_tb_pol/test_venv/lib/python3.\
6/site-packages/pkg_resources/__init__.py", line 565, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File
"/home/data/verification/users/millsd/workspace/uvm_mk_tb_pol/test_venv/lib/python3.\
6/site-packages/pkg_resources/__init__.py", line 2631, in load_entry_point
    return ep.load()
  File
"/home/data/verification/users/millsd/workspace/uvm_mk_tb_pol/test_venv/lib/python3.\
6/site-packages/pkg_resources/__init__.py", line 2291, in load
    return self.resolve()
  File
"/home/data/verification/users/millsd/workspace/uvm_mk_tb_pol/test_venv/lib/python3.\
6/site-packages/pkg_resources/__init__.py", line 2297, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File
"/home/data/verification/users/millsd/workspace/uvm_mk_tb_pol/test_venv/lib/python3.\
6/site-packages/hdl_checker/__init__.py", line 29, in <module>
    from hdl_checker.utils import ON_WINDOWS
  File
"/home/data/verification/users/millsd/workspace/uvm_mk_tb_pol/test_venv/lib/python3.\
6/site-packages/hdl_checker/utils.py", line 35, in <module>
    import six
ModuleNotFoundError: No module named 'six'
```

I am using CentOS 7.

DeeeeLAN, 2020-08-06:
doing a `pip install -U six` resolved the issue. The package needs to be added to the list of requirements so it gets installed automatically along with the other dependencies.

suoto, 2020-08-09:
This has been fixed but not released to PyPI, will make a new release from latest, thanks for the reminder!

closed 2020-08-10 by suoto