suoto/hdl_checker #88: pygls.features no longer exists

benjaminmordaunt, 2021-03-18:
Freshly installing hdl_checker on Python 3.9 produces the following error:

`ModuleNotFoundError: No module named 'pygls.features'`

Looking through the current pygls source, it seems as though `features.py` was removed in a recent version?

suoto, 2021-03-20:
Yes, it seems like it was removed indeed, I'll have a look at that. In the meantime, if you don't need pygls 0.10.0 or newer, you can use version 0.9.1 (the last version before this change):
```
pip uninstall pygls
pip install pygls==0.9.1
```

closed 2021-04-11 by suoto

Freed-Wu, 2023-06-30:
@suoto Can you update the code to adapt new pygls? I have many language servers use latest pygls which have conflict with hdl_checker.

Freed-Wu, 2023-06-30:
I try to fix this issue, however it still have some conflicts:

```diff
diff --git a/hdl_checker/lsp.py b/hdl_checker/lsp.py
index fae18ad..4992229 100644
--- a/hdl_checker/lsp.py
+++ b/hdl_checker/lsp.py
@@ -23,19 +23,17 @@ from os import path as p
 from tempfile import mkdtemp
 from typing import Any, Callable, Iterable, List, Optional, Set, Tuple, Union
 
-from pygls.features import (
-    DEFINITION,
-    HOVER,
+from pygls.server import LanguageServer
+from lsprotocol.types import (
+    TEXT_DOCUMENT_DEFINITION,
+    TEXT_DOCUMENT_HOVER,
     INITIALIZE,
     INITIALIZED,
-    REFERENCES,
+    TEXT_DOCUMENT_REFERENCES,
     TEXT_DOCUMENT_DID_CHANGE,
     TEXT_DOCUMENT_DID_OPEN,
     TEXT_DOCUMENT_DID_SAVE,
     WORKSPACE_DID_CHANGE_CONFIGURATION,
-)
-from pygls.server import LanguageServer
-from pygls.types import (
     ClientCapabilities,
     Diagnostic,
     DiagnosticSeverity,
@@ -570,17 +568,17 @@ def setupLanguageServerFeatures(server: HdlCheckerLanguageServer) -> None:
     ) -> None:
         self.onConfigUpdate(settings)
 
-    @server.feature(HOVER)
+    @server.feature(TEXT_DOCUMENT_HOVER)
     def onHover(self: HdlCheckerLanguageServer, params: HoverParams) -> Optional[Hover]:
         return self.hover(params)
 
-    @server.feature(REFERENCES)
+    @server.feature(TEXT_DOCUMENT_REFERENCES)
     def onReferences(
         self: HdlCheckerLanguageServer, params: ReferenceParams
     ) -> Optional[List[Location]]:
         return self.references(params)
 
-    @server.feature(DEFINITION)
+    @server.feature(TEXT_DOCUMENT_DEFINITION)
     def onDefinition(
         self: HdlCheckerLanguageServer, params: TextDocumentPositionParams
     ) -> Optional[List[Location]]:
diff --git a/hdl_checker/server.py b/hdl_checker/server.py
index 7c1ee94..99f1a4b 100755
--- a/hdl_checker/server.py
+++ b/hdl_checker/server.py
@@ -221,7 +221,7 @@ def run(args):
 
     if args.lsp:
         stdin, stdout = _binaryStdio()
-        server = lsp.HdlCheckerLanguageServer()
+        server = lsp.HdlCheckerLanguageServer("hdl_checker", version)
         lsp.setupLanguageServerFeatures(server)
         server.start_io(stdin=stdin, stdout=stdout)
     else:
diff --git a/setup.py b/setup.py
index 4730e16..3f0ad16 100755
--- a/setup.py
+++ b/setup.py
@@ -59,7 +59,7 @@ setuptools.setup(
                                      'future>=0.14.0',
                                      'futures; python_version<"3.2"',
                                      'prettytable>=0.7.2',
-                                     'pygls==0.9.1',
+                                     'pygls>0.9.1',
                                      'requests>=2.20.0',
                                      'six>=1.10.0',
                                      'tabulate>=0.8.5',
```

commit 03e179e6 (2021-04-11): Fixed pygls version to 0.9.1 (closes #88)

Minor changes in the docker stuff