suoto/hdl_checker #93: NeoVim Native LSP Configuration errors

SethGower, 2021-07-21:
# Issue details

The configuration in the readme for the native `nvim` LSP client doesn't seem to be correct. It should be `lspconfig`, not `nvim_lsp` (this requires the [`lspconfig`](https://github.com/neovim/nvim-lspconfig) plugin to be installed, but this is the normal requirement). I also feel like the current directory is a better fall back when a proper root can't be found as opposed to using the home dir. Using home dir leads to _exceedingly_ long load times. 

I personally use `.hdl_checker.config` as root, and if it can't find, it uses current dir. This is simply because at work I deal with projects that are comprised of multiple repos, but need to see files across repos, so I have the `.hdl_checker.config` in the parent dir to the repos. 

I recommend something like this
```lua
local nvim_lsp = require'lspconfig'
if not nvim_lsp.hdl_checker then
  require'lspconfig/configs'.hdl_checker = {
    default_config = {
    cmd = {"hdl_checker", "--lsp", };
    filetypes = {"vhdl", "verilog", "systemverilog"};
      root_dir = function(fname)
        return nvim_lsp.util.find_git_ancestor(fname) or util.path.dirname(fname)
      end;
      settings = {};
    };
  }
end
```
This is my config, from my `init.lua` in my [dotfiles](https://github.com/SethGower/dotfiles/blob/main/.config/nvim/init.lua#L244)

```lua
local lspconfig = require('lspconfig')
local configs = require('lspconfig/configs')
local util = require('lspconfig/util')

if not lspconfig.hdl_checker then
  configs.hdl_checker = {
    default_config = {
      autostart = false, -- disabled auto start since I am giving rust_hdl a shot
      cmd = {"hdl_checker", "--lsp"};
      filetypes = { "vhdl" };
      root_dir = function(fname)
        return util.root_pattern('.hdl_checker.config')(fname) or util.path.dirname(fname)
      end;
      settings = {};
    };
  }
end
```

(none of the system stuff is really necessary, except for maybe my `nvim` version)
- `nvim -v`
```
NVIM v0.6.0-dev
Build type: RelWithDebInfo
LuaJIT 2.1.0-beta3
Compilation: /usr/bin/cc -g -O2 -fdebug-prefix-map=/build/neovim-MntHic/neovim-0.5.0+ubuntu2+git202107121802-eece0735f-d569569c9=. -fstack-protector-strong -Wformat -Werror=format-security -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=1 -DNVIM_TS_HAS_SET_MATCH_LIMIT -O2 -g -Og -g -Wall -Wextra -pedantic -Wno-unused-parameter -Wstrict-prototypes -std=gnu99 -Wshadow -Wconversion -Wmissing-prototypes -Wimplicit-fallthrough -Wvla -fstack-protector-strong -fno-common -fdiagnostics-color=auto -DINCLUDE_GENERATED_DECLARATIONS -D_GNU_SOURCE -DNVIM_MSGPACK_HAS_FLOAT32 -DNVIM_UNIBI_HAS_VAR_FROM -DMIN_LOG_LEVEL=3 -I/build/neovim-MntHic/neovim-0.5.0+ubuntu2+git202107121802-eece0735f-d569569c9/build/config -I/build/neovim-MntHic/neovim-0.5.0+ubuntu2+git202107121802-eece0735f-d569569c9/src -I/build/neovim-MntHic/neovim-0.5.0+ubuntu2+git202107121802-eece0735f-d569569c9/.deps/usr/include -I/usr/include -I/build/neovim-MntHic/neovim-0.5.0+ubuntu2+git202107121802-eece0735f-d569569c9/build/src/nvim/auto -I/build/neovim-MntHic/neovim-0.5.0+ubuntu2+git202107121802-eece0735f-d569569c9/build/include
Compiled by buildd@lgw01-amd64-049

Features: +acl +iconv +tui
See ":help feature-compile"

   system vimrc file: "$VIM/sysinit.vim"
  fall-back for $VIM: "/usr/share/nvim"

Run :checkhealth for more info
```

I can eventually make a pull request with this fix. But it might be a bit, since I am busy with work/moving soon.

SethGower, 2021-07-21:
additionally, it should specify you need to call some form of 
```lua
require('lspconfig').hdl_checker.setup()

suoto, 2021-07-24:
Good catch :)

PR #94 (2021-07-24): updates nvim-lsp section of readme
Fixes #93
closed 2021-07-24 by suoto