suoto/hdl_checker #109: not support Python 3.12

laciel-chin, 2024-03-20:
# Issue details

* Python version used
Python 3.12.2 (tags/v3.12.2:6abddd9, Feb  6 2024, 21:26:36) [MSC v.1937 64 bit (AMD64)] on win32

* HDL Checker log output
File "....\versioneer.py", line 412, in get_config_from_root
          parser = configparser.SafeConfigParser()
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'configparser' has no attribute 'SafeConfigParser'. Did you mean: 'RawConfigParser'?

* reason

Several names deprecated in the [configparser](https://docs.python.org/3.12/library/configparser.html#module-configparser) way back in 3.2 have been removed per https://github.com/python/cpython/issues/89336:

[configparser.ParsingError](https://docs.python.org/3.12/library/configparser.html#configparser.ParsingError) no longer has a filename attribute or argument. Use the source attribute and argument instead.

[configparser](https://docs.python.org/3.12/library/configparser.html#module-configparser) no longer has a [SafeConfigParser](https://docs.python.org/3.12/library/configparser.html#configparser.ConfigParser) class. Use the shorter ConfigParser name instead.

[configparser.ConfigParser](https://docs.python.org/3.12/library/configparser.html#configparser.ConfigParser) no longer has a readfp method. Use [read_file()](https://docs.python.org/3.12/library/configparser.html#configparser.ConfigParser.read_file) instead.

0dminnimda, 2024-06-24:
This would likely be closed by #110

bellaz89, 2025-01-14:
Any update on this?

suoto, 2026-09-23:
Fixed in v0.7.5. The crash came from the bundled `versioneer.py` calling the removed `configparser.SafeConfigParser`; v0.7.5 migrates to setuptools-scm and drops the legacy code, adding support for Python 3.11–3.13.

closed 2026-09-23 by suoto