suoto/hdl_checker #114: Installation fails with Neovim's lspconfig, as well as via pipx

ObjectsCountries, 2025-04-02:
When trying to install this package with either pipx or Neovim, I get the following error:

> AttributeError: module 'configparser' has no attribute 'SafeConfigParser'. Did you mean: 'RawConfigParser'?

More details can be found with Neovim via Mason:

```
          × python setup.py egg_info did not run successfully.
          │ exit code: 1
          ╰─> [26 lines of output]
              /tmp/pip-install-0m8n5shr/hdl-checker_ce8fc1f730b9436e84f9e2e049dfea9b/versioneer.py:485: SyntaxWarning: invalid escape sequence '\s'
                LONG_VERSION_PY['git'] = '''
              Traceback (most recent call last):
                File "<string>", line 2, in <module>
                  exec(compile('''
                  ~~~~^^^^^^^^^^^^
                  # This is <pip-setuptools-caller> -- a caller that pip uses to run setup.py
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                  ...<31 lines>...
                  exec(compile(setup_py_code, filename, "exec"))
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                  ''' % ('/tmp/pip-install-0m8n5shr/hdl-checker_ce8fc1f730b9436e84f9e2e049dfea9b/setup.py',), "<pip-setuptools-caller>", "exec"))
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                File "<pip-setuptools-caller>", line 34, in <module>
                File "/tmp/pip-install-0m8n5shr/hdl-checker_ce8fc1f730b9436e84f9e2e049dfea9b/setup.py", line 43, in <module>
                  version                       = versioneer.get_version(),
                                                  ~~~~~~~~~~~~~~~~~~~~~~^^
                File "/tmp/pip-install-0m8n5shr/hdl-checker_ce8fc1f730b9436e84f9e2e049dfea9b/versioneer.py", line 1473, in get_version
                  return get_versions()["version"]
                         ~~~~~~~~~~~~^^
                File "/tmp/pip-install-0m8n5shr/hdl-checker_ce8fc1f730b9436e84f9e2e049dfea9b/versioneer.py", line 1406, in get_versions
                  cfg = get_config_from_root(root)
                File "/tmp/pip-install-0m8n5shr/hdl-checker_ce8fc1f730b9436e84f9e2e049dfea9b/versioneer.py", line 412, in get_config_from_root
                  parser = configparser.SafeConfigParser()
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
              AttributeError: module 'configparser' has no attribute 'SafeConfigParser'. Did you mean: 'RawConfigParser'?
```

I am running EndeavourOS, with Python 3.13.2, Neovim version 0.11.0 and pipx version 1.7.1.

ChausseBenjamin, 2025-05-04:
Same here on arch...

ilnerdchuck, 2025-11-09:
> [mason-org/mason.nvim#1916](https://github.com/mason-org/mason.nvim/issues/1916)
> 
> I found this, might be helpful form someone here.

Same on Fedora, solved with this

suoto, 2026-09-23:
Fixed in v0.7.5. The install-time `SafeConfigParser` error came from the bundled `versioneer.py`; v0.7.5 removes it (migrated to setuptools-scm), so installation via pipx / Neovim's lspconfig no longer fails.

closed 2026-09-23 by suoto